packofparts/frc2024 #2: Finishing T-Shirt Cannon

ArvickC, 2023-06-26:
Priority: High

Started in pre-season, would be good for school events/games

BilledGarlic44, 2023-06-27:
Need to decide on air flow design, valve type (ball, butterfly, sprinkler), and barrel count as well as drive base

aditya0shah, 2023-06-29:
Electrical can work on pneumatics for tshirt cannon

BilledGarlic44, 2023-07-07:
Ask local fire departments for "retired" or "retiring" SCUBA/SCBA tanks for main air storage

BilledGarlic4, 2023-07-22:
Here are a few design requirements i've gathered from reading a bunch of threads and whatnot. Feel free to add any reqs you feel are necessary, or discuss/critique any of my reqs

Possibly Multi Barrel using Geneva drive (https://www.youtube.com/watch?v=1lyWywC_z4o)
Use Non-PVC barrels ideally (don' want a pipe bomb)
2 SCUBA tanks for air storage. (feed into secondary storage directly behind cannon)
Long barrel
atlest 3/4" to 1" output hose/tubing (garden hose)
Main tank to holder/accumilator tank has regulator in between.
Gaskets/sealing mech for multibarrel is nice but not necessary
Solenoid ball valve or Solenoid butterfly valve
analog pressure sensor to know when to fire w/ shuffleboard PSI reading
add throughbore encoder or limit switch for articulation limit
ensure top notch branding
ensure easy and safe fixability and usability (High pressure air is dangerous!!!!)

A great place to get some more info: https://www.chiefdelphi.com/t/building-a-t-shirt-cannon-robot/414536/50

ArvickC, 2023-10-30:
Removed from pre season project list