packofparts/frc2024 #6: Maintain Riptide

ArvickC, 2023-06-26:
Priority: High

Swap out the arm for arm v2, rewire, and convert to mk 4. Requires mech to build new arm and attach new swerve modules.

aditya0shah, 2023-06-27:
Need mechanical sub team to make and attach arm v2 
@FRC-1294/mechanics

Ants1the1, 2023-07-20:
Build subteam has started to assemble the Mk4 modules but ran into issues finding usable bolts. At the end of 7/19 meeting we found the bolts we needed.
Going into 7/26 meeting we should be able to fully assemble modules and hand off to electrical to wire.

Ants1the1, 2023-08-03:
> Build subteam has started to assemble the Mk4 modules but ran into issues finding usable bolts. At the end of 7/19 meeting we found the bolts we needed.
> Going into 7/26 meeting we should be able to fully assemble modules and hand off to electrical to wire.

The modules are mechanically done and ready for electrical to wire and test. 

We are wanting to iterate the wooden spacers to help with routing the wires and holding the sparks in place

aditya0shah, 2023-08-09:
@FRC-1294/electronics are cancoders attached?

FBurkelwsd, 2023-08-11:
@TechTaco1 @Ants1the1 @ArvickC  have been cleaning up wire routing, wire connections, updated swerve modules to mk4, and are re configuring placement of some electrical components on the bottom robot board.

SanchitArora44, 2023-08-26:
Arm V2's outside tube is nearly finished, and dual roller intake plates are cut out. Once arm is finished (ideally some time next week), will begin to replace battered tubes in drivebase one by one.

aditya0shah, 2023-09-03:
Arm v2 has been wired but the wires are yet to be neatly organized. We are waiting to 3d print a mount that Sanchit's designed to mount deutch connectors to arm.  Claw wiring has not been started yet and we need to attach an energy chain to arm. Once we know if it's a bag motor or neo for the claw we can begin wiring that. In the case that it is a neo motor @FRC-1294/electronics needs to remember to remove the falcon and mount a spark on the telescopic (inside) arm.

adshah2, 2023-09-14:
Robot is pretty much all wired up. Mech has slight telescopic arm issue to fix and elec has to finish routing cables but robot is nearly ready for testing.

Ants1the1, 2023-10-05:
Extension is working well enough to test on and hopefully compete with.

Mechanical To-do: 
1. Finish mounting new neo motor and gearbox
2.mount bearings to top of bearing block
3. Finish Red Rear bumper mounts (raise mounts either 0.18 ish inches or 0.25 inches)
4. wrap with bumper fabric
5. Install "studs" into the frame for red bumper mounts
6.Bumper numbers (paint?)
7. Bolt check the swerve modules

FBurkelwsd, 2023-10-06:
Paint is on the way, will be here on Tuesday.

aditya0shah, 2023-10-10:
Electrical to do: 
1. Install encoder extension (+ joiner board)/power/sensor wire extensions to claw and remount spark to base of board
2. clean up any loose wires/connectors on board
3.  If canivore has come, route falcons/cancoders on seperate CAN bus (or at least just cancoders for now)
4. Organize all things we're taking to comp

adshah2, 2023-10-25:
Its been fun...
but onwards

closed 2024-01-11 by adshah2