packofparts/frc2024 #8: Experiment with different electrical connectors

ArvickC, 2023-06-26:
Priority: Mid

Want to test out different connectors mainly Powerpole/Wago to see if better for our needs.

aditya0shah, 2023-06-29:
Add to the list solder seal wire, especially for canBUS

We must also identify the root causes of our electrical connector issues to figure out our best solutions

aditya0shah, 2023-07-14:
Ordered solder seal wire (100 pack of 20 awg and 100 pack of 12 awg)

Yet to order Anderson PowerPoles

aditya0shah, 2023-07-21:
Current plan: 

test solder seal wire, if effective and easy enough for repairability, we can use that for CAN (else will be either powerpole/deutch).

Options for power are Andersons PowerPole/Deutch

PowerPole pros:
- cheap
- many teams use (reparability in comp)
cons: 
- not as durable as deutch
- have to completely change entire system

Deutch pros:
- durable
- current system
cons: 
- bulky
- expensive

Current tasks: 

Test PowerPole:
- test ease of usability
- test durability
- get @TJScher opinion on them

Research Deutch:
- Find which products to buy
- total cost of ordering deutches for this year (include pins/sockets)

TechTaco1, 2023-07-27:
Unable to thoroughly test as we didn't have the right crimpers for powerpoles.

tested solder seal a little bit:
-strong, only broke because of cold solder and lots of folding 
-permanent
-when need to replace need to cut off alot of the wire

we need to test solder seal more as i don't feel good with the test we did (still surprisingly strong)

aditya0shah, 2023-07-27:
@TechTaco1 how much wire was cut off when cutting the solder seal wire. Would you be able to estimate how many times you can do that on a Spark CAN bus for example? What are your thoughts on its usability for us especially if we do have to cut the solder seal wire multiple times (even if we're trying hard to limit our cutting)?

TechTaco1, 2023-08-03:
The amount of wire that is needed to cut off is a lot. about 0.7-0.8 inches on both wires. also, after testing more with a heat gun, it results in a cold solder and burnt heat shrink.

TechTaco1, 2023-08-03:
## Powerpoles and deutsch:

both reliable connectors,
both are able to be mounted so weight doesn't matter,
both easy to take apart

### Powerpoles:
**pros**

- smaller
- used by other teams
- fast to put together 
- easy to teach
- easy to see what wires go where at a glance
- cheaper

**cons**

- connector is less durable (not by much)
- have to change entire system

### Deutsch:
**pro**

- very durable
- using them for years

**cons**
- Harder to teach
- expensive
- big
- not used by other teams

aditya0shah, 2023-08-09:
Ok great. I'd like to do a little testing myself when I am back but I do think powerpoles would be a viable option for us.

@FRC-1294/electronics At todays meeting compile a list for the connectors and quantities we would need (including pins) and calculate a total price if we were to transfer over to powerpoles.

TechTaco1, 2023-08-10:
[image]

this is based off the numbers that we were going to by for deutsch

Anderson                      Deutsch                           how many of each
Red, Black, White      4 pin size 12               200 red, 200 black, 200 white and 600 size 30 amps

wygrangea, 2023-08-14:
@TechTaco1 I walked through our robot today with our Tech leads, We estimated up to 15 motors on a robot, which require about 14 Andersons each: 2 green, 2 yellow, 4 red, 4 black, 2 white. This includes tying into the pdh but doesn't include any sensors you might use. So I think if you convert you could safely scale back your order slightly. You're missing a few key things, I recommend getting an organizer for them, and you could start with the kit below that includes the removal tool, retention pins and roll pins which are good to have. Then add to it from what else you order. 
[image]

TechTaco1, 2023-08-17:
120 red
120 Black
70 white
170 16-20 pins
270 12-14 pins
1 crimper
50 green
50 yellow

this is everything divided by 3 so this should be good for at least for 1 season.

wygrangea, 2023-08-17:
Got it. Added in retention clips and the release tool, plus tweaked the numbers on the contacts (pins) so they coordinate better with the number of housings ordered.

ArvickC, 2023-08-26:
@FRC-1294/electronics chose Andersons as the connectors, and everything (included tools) has arrived. Should I close this issue?

wygrangea, 2023-08-26:
I think so. If more needs to be ordered or changes made We can open another issue I think.

closed 2023-08-26 by ArvickC