packofparts/frc2024 #14: April tag localization finalization

ArvickC, 2023-06-27:
Priority: Mid

AadityaAgni, 2023-07-14:
For anyone that wants to research this beforehand, especially the pose estimation system, here are a couple links that you can use:

[Limelight Docs](https://docs.limelightvision.io/en/latest/apriltags_in_3d.html): Limelight primarily communicates through NetworkTables, which is how we will retrieve the botpose estimations from april tags. Note that we use the value with the origin at the bottom left corner to work with the Field2d convention. Vision is one essential measurement in pose estimation as a whole, since we can find where the bot is in a global context unlike odometry, and it can account for our odometry drift issues.

[Swerve Drive Pose Estimator](https://first.wpi.edu/wpilib/allwpilib/docs/release/java/edu/wpi/first/math/estimator/SwerveDrivePoseEstimator.html): This is the class we use to combine readings from both vision and odometry. It essentially just does our job for us, doing all the necessary calculations for combining data with different reliabilities and timestamps into one convenient pose2d. Something to pay attention to is the concept of measurement standard deviations you can pass in when updating a value, allowing us to "trust" a certain value more or less by adjusting that.

If you're curious as to my (partial) implementation of Pose Estimation as a reference, you can see last year's repo, where there is a folder dedicated to pose estimation. Make sure to look at the one for limelight, as we no longer use PhotonVision. Note: The limelight version is probably wrong because I never got to test it, but it should theoretically work(although in this line of work that basically means nothing).

FBurkelwsd, 2023-09-26:
https://docs.limelightvision.io/en/latest/

Are we testing any of these pipelines? or are they relevant to our needs? What is the system, other than limelight we could use?

AadityaAgni, 2023-09-27:
Right now we have a set of options and firmware to use. Here's a quick rundown

Limelight: This is a closed, out-of-the-box solution that solely functions with Limelight hardware. For off season comps we are going with this as its the easiest to implement as we focus on the features that go with vision, such as autonomous routines. The benefit of this is that it takes the least time to set up. The harm is that we can't improve it very much in the future because its such a closed solution.

PhotonVision: An open source, objectively better solution that we are going to switch to in the future. The benefit of this is that we can get much better hardware than limelight for much cheaper, as well as improve it greatly from our previous limelight setup. The hardware setup I was hoping to use with photonvision is an orangepi, since an orangepi setup is several times faster and more efficient than a limelight at a cheaper cost. The harm is that it takes longer to set up, which is why we are saving it for later

TLDR: We(along with many many other teams) have begun moving away from the closed solution of limelight, as we no longer needs what the limelight used to offer that was so appealing: an easy led reflective tape solution. In the short term, limelight is easier to set up, but in the long term, photonvision is already much easier to grow and has much more potential for the future.

FBurkelwsd, 2023-09-27:
So for limelight, all of the code integrations/pipelines listed are no help to open up the possibilities?

AadityaAgni, 2023-09-27:
Yup! Vision has 3 main pipelines that we could use: 

April Tags: For pose estimation, and photon does this much better
Reflective Tape: The most compelling part of the limelight during its first years was this because it was the first reflective tape solution out of the box. Reflective tape has been discontinued by FRC for the next season, so we can't use this anymore, making the best part of limelight completely worthless. Photon's equivalent to this is a colored shape pipeline, which we can continue using because it isn't dependent on tape.
Object Detection: We don't use this, but both LL and Photon have this, although LL does have coral integration, which Photon doesn't have. However, this is a feature we haven't even planned on using just yet, so this shouldn't hold us back.

AadityaAgni, 2023-10-02:
I don't think this issue is finished as we are still testing/working on this. Not to mention there is still much work to be done to switch to photonvision during offseason.

AadityaAgni, 2023-10-05:
Here are some MiniPC Resources

[Setup Guide and Robustification](https://docs.google.com/document/d/1lOSzG8iNE43cK-PgJDDzbwtf6ASyf4vbW8lQuFswxzw/edit)

[Testing For Different Cameras:
](https://docs.google.com/document/d/1AU8rK364pTvNaS69ZJ6ZdovyLU_sL_PJ9WSHSJIVoqk/edit#)

[Chief Delphi Thread
](https://www.chiefdelphi.com/t/a-high-performance-apriltag-solution-beelink-mini-pc-photonvision-guide/419142)
From what I can tell, the best two options we have are an OV9281 and an AR0144, although a lifecam should be ok

Arducam USB OV9281 Global shutter camera: https://www.amazon.com/Arducam-Distortion-Microphones-Computer-Raspberry/dp/B096M5DKY6?th=1

AR0144 camera: https://www.amazon.com/Global-Shutter-Monochrome-Cameras-Windows/dp/B089QFHHZL?th=1

AadityaAgni, 2023-10-27:
Now that the limelight 2+ has broken, that leaves us with two main options for vision during the offseason:

1. MiniPC: Most Likely/Best Option because this is faster at a higher resolution than the Limelight 2+. However, there is more work needed to build/wire as well as set it up than a Limelight, but I would argue that it is worth it. I posted setup instructions and which camera we should pair with a MiniPC above. In particular, an Arducam OV9281 is the best option.

3. Limelight 3: Currently, there is still no version of PhotonVision for the Limelight 3, but there may be a potential workaround. I was able to get PhotonVision to boot on a Limelight 3 with a great degree of difficulty, but more testing is needed before I can even assert that this is an option.

AadityaAgni, 2023-10-30:
Looks like we have purchased the cameras, so I'm going to post a couple of updates about the actual process of setting up multi-cam vision with the mini PC is going to be. 

**What the setup is looking like**: Because photon-vision can't have two of the same camera, the vision setup I'm aiming for is a mini PC and an Arducam OV9281 + AR0144 camera setup, which seems to be the best options based on the research I did. Preferably, we would have one camera on the back and one on the front at the top of whatever robot we build next season for peak performance.

1. Electrically: @adshah2, there are a couple of questions I had.
 - Could you look into the process of actually setting up a mini PC electrically? I posted a couple of guides above which should address this. There might be a couple of components that we don't have, so knowing in advance might be helpful. In particular, we may have to find wires long enough to have the cameras be at the top of the robot.
- 2 Cameras will understandably be absolutely terrible for CAN utilization, so I just wanted to know whether its even a feasible option.

3. Design: @SanchitArora44 the design subteam might have to create a mount for the mini-pc and cameras
4. Software: We will be setting up a mini-pc with a Microsoft lifecam as a baseline POC just to ensure that everything is functional and whatnot. Also, we need to deal with the logistics of actually having two cameras at once in code, which shouldn't be that bad.

AadityaAgni, 2023-11-17:
MiniPC Update: Finished installing Ubuntu and Photonvision, and the dashboard seems to be working. Next Steps:

- Figure out static IP addresses
- Test Cameras
- Figure out ssh by hostname and not IP, which isn't working for some reason. If anyone is more experienced with that type of stuff, it would be mucho helpful. 
- Performance tests(the fun part)
- Once all that is done, we need to test it on the robot with the network switch and mounted cameras. 

[I'm currently working off of this guide](https://docs.google.com/document/d/1lOSzG8iNE43cK-PgJDDzbwtf6ASyf4vbW8lQuFswxzw/edit)

AadityaAgni, 2023-11-18:
MiniPC Update Day 2: Had to go slightly off script with setting up static IP addresses, but so far I think it works. I also tested with one of our cameras and photonvision successfully detected it! More detailed updates/todo:

Finished:
- Theoretically figured out static IP address through wifi(can apply through ethernet later)
- Tested Camera(works)

TODO:
- Figure out ssh by hostname and not IP. I tried for over an hour using avahi-daemon, but I have no clue what I'm doing and my sanity is decaying. My best guess is that this is a problem with my home wifi, and with the smaller LAN of the Robot, this should be easier to do. Worst case scenario we just access it by IP.
- Performance Testing
- Testing on Robot Network

Final Thoughts: MiniPC is hard. With Limelight we could get away with flashing and then just plug-and-play but minipc requires painstaking labor to set up the network.

closed 2024-04-17 by ArvickC