packofparts/frc2024 #27: Attendance

jjalla, 2023-07-13:
We have a working Google form. We are going to test it in the August off-season meetings and officially use it when the preseason starts in September.

ArvickC, 2023-07-23:
Assuming that [this](https://docs.google.com/forms/d/10CTm7m7Kta8Nx73PUj1XMaZHH2FHUv7-mOverXgoH6E/edit?pli=1) is the link to the attendance form. Would be good to add a name/email section, or for more security/less "attending for a friend" collecting email data, and limiting the response to 1 per user

jjalla, 2023-07-23:
Yes not sure why we don't have a name/email section. Limiting the response to 1 per user is a good idea I think for more security we are also going to only open the form during the first ten minutes of the meeting so it limits people passing around the form.

Eshan-Feroz, 2023-07-25:
Edited the form to collect name, email addresses and the current date and
disabled the link "Submit another response" at the end of the form. Since
we were planning to reuse this form for every team meeting, wouldn't
limiting the response to 1 per user be a problem?

Good idea to open the form for the first 10 mins.

Another option is to create individual forms for each team meeting/monthly
forms, which could actually be easier to track.
Let me know what you think.

closed 2023-09-25 by ArvickC