packofparts/frc2024 #39: Order Deutch Pins

aditya0shah, 2023-07-20:


aditya0shah, 2023-07-21:
@Benstromseth @wygrangea 
Let's hold off on ordering these until we're certain we prefer Deutch for upcoming season. 
Ms. Dooley, could you bring PowerPoles (of various sizes) to next meeting for testing? 
@FRC-1294/electronics please experiment with powerpoles and document ease of putting together/durability

wygrangea, 2023-07-21:
Sounds good @aditya0shah . I don't know that I will be able to bring much to the next meeting because I am not certain I'll be able to get into the shop. I will be able to bring the right size for the motors though, I think they use 16-18 gauge wire. I have those in my garage for FTC.

BilledGarlic44, 2023-07-21:
For Deutsch atleast, motors use 12 gauge pins/sockets. Not sure if it's a 1:1 transfer in terms of size when it comes to powerpoles, but just wanted to let you know

closed 2023-08-26 by aditya0shah