packofparts/frc2024 #41: Order Deutch Connectors

aditya0shah, 2023-07-20:
Order the actual Deutch connectors themselves for a restock for the upcoming season.
@TechTaco1 please comment which connectors are on that order list we made last meeting.

TechTaco1, 2023-07-21:
4 pin size 12 x100 more male/female
2 pin size 12 x50 more male/female
4 pin size 20 12 x30 more male/female
2 pin size 20 12 x75 more male/female

TechTaco1, 2023-07-27:
where to buy Deutsch connectors 

[image]

DTP connectors (size 12 connectors)

DTP06-4S mouser electronics/Digi-key electronics (cannot buy from future as min order is 500)
DTP04-4p Future Electronics
DTP04-2p Future Electronics
DTP06-2s Future Electronics

DTM (size 20 connectors)

DTM04-4p Future electronics
DTM06-4s Future electronics 
DTM04-2p from distributer 
DTM06-2s Future electronics 

after we decide what connectors, we will use we can order from here.

TechTaco1, 2023-08-11:
[image]

[image]

[image]

wygrangea, 2023-08-14:
Next step: meet to inventory and make a final connector decision with Ben. plan for 8/14 6:30pm

closed 2023-08-26 by aditya0shah