packofparts/frc2024 #46: Rev team sponsorship

aditya0shah, 2023-07-22:
https://www.revrobotics.com/team-sponsorship/
https://www.revrobotics.com/frc-sponsorship-application/

Sponsorship from Rev Robotics.
Must be completed by a mentor, or at least have a mentor in contact info (theres a checkbox that says the form was filled out by someone 18+ who is primary point of contact for team)

wygrangea, 2023-07-24:
This has been assigned to many people now. If you want someone to do it please ask them, or send a note when assigning. I will wait to see if I actually need to do anything for this. I'm willing, but just assigning me and a bunch of others doesn't work for me.

FBurkelwsd, 2023-07-25:
Extracted from application: 

Upload supporting documentation (e.g. judging materials, robot photos, or CAD renders) from the previous 3 seasons

@FRC-1294/pr

FBurkelwsd, 2023-07-25:
> Extracted from application: 
> 
> Upload supporting documentation (e.g. judging materials, robot photos, or CAD renders) from the previous 3 seasons
> 
> 
> @FRC-1294/pr 

"Tell us about anything else that makes your team unique. What sets your team apart from others?"

FBurkelwsd, 2023-07-25:
> > Extracted from application: 
> > 
> > Upload supporting documentation (e.g. judging materials, robot photos, or CAD renders) from the previous 3 seasons
> > 
> > 
> > @FRC-1294/pr 
> 
> "Tell us about anything else that makes your team unique. What sets your team apart from others?"

I can fill in the rest, wouldn't mind someone to review as I typed to minimize error. 

@c

shaankal, 2023-08-11:
Currently Creating new file in drive, which will include docs necessary for submission.

shaankal, 2023-08-11:
Also drafting "Tell us about anything else that makes your team unique. What sets your team apart from others?" , can send for thoughts when done

FBurkelwsd, 2023-09-06:
https://docs.google.com/document/d/1jqH_S2aSn4AUnkwKTzJKJwedB7kwDTKbwDUHRGk0ngU/edit
glad there are some answers towards the bottom of this document. will create a new issue for the other sponsors as well. I just dont want to have to do this all alone again.

FBurkelwsd, 2023-09-09:
> > > Extracted from application:
> > > Upload supporting documentation (e.g. judging materials, robot photos, or CAD renders) from the previous 3 seasons
> > > @FRC-1294/pr
> > 
> > 
> > "Tell us about anything else that makes your team unique. What sets your team apart from others?"
> 
> I can fill in the rest, wouldn't mind someone to review as I typed to minimize error.
> 
> @ArishaG @shaankal @Eshan-Feroz @pk2107 @jjalla

Pack of Parts lives our mission statement of inspiring every generation with hands on stem literacy, team organized robotics camps, laser cut stem kits, and empowering students

FBurkelwsd, 2023-09-09:
Very student led" from ben and laura

closed 2023-09-14 by Eshan-Feroz