packofparts/frc2024 #47: Farmers Market

aditya0shah, 2023-07-22:
We will be going to the Farmer's Markets on August 16th and September 6th.
Still need to contact Deb.

BilledGarlic4, 2023-07-23:
August 16th seems like a tough deadline to meet for arm V2 and new intake considering our current progress. Is the robot something we absolutely need to have finished by the farmers market or is it fine if it's just a drivebase moving around?

Eshan-Feroz, 2023-07-23:
That should be fine considering main point of farmers market is for people
to get interested. A drivebase is enough.

Eshan-Feroz, 2023-07-25:
Assigned @ArishaG

FBurkelwsd, 2023-08-01:
Things we want to design/compile:
Materials needed for set up
Directors and member input on content 
Transport 
Student leaders 
Goals

Eshan-Feroz, 2023-08-12:
Compiled a list of materials for set up found below, waiting for board/mentor input: 
https://docs.google.com/document/d/1psVPX8ywlVnN_pbQnAhHLAIRNGYBlhEZiORjKzLEacI/edit

Will create and send out a mentor/board sign up form and a team volunteer sign up next week
@Ants1the1 (Alex T.) agreed to transport the robot.

ArvickC, 2023-08-26:
The farmers market has finished. Should I close this issue? Or would we like to have it open for further farmer market/outreach events.

closed 2023-09-06 by jjalla