packofparts/frc2024 #48: Constitution and Bylaws reviewed

aditya0shah, 2023-07-22:
https://docs.google.com/document/d/1YV2juxjXIEoMS_Da28Iz4OFE_OaigK4_hUW4AjzMi5w/edit?usp=sharing is the link to the constitution

https://docs.google.com/document/d/1adjQDJysnLXXXWAJ5RRWyyAXN_MHf6caftYLt-t7cJg/edit?usp=sharing is the link to the bylaws

Lets have a first comment and proposed revisions before August 9th.

geeknoid, 2023-07-22:
I left a bunch of comments in the docs. I think we should assign someone to start from these documents and create a fresh update that we can iterate over. Once we're happy with the updated documents, we put the constitution up for a vote, and assuming it passes, then we paste the new content into the existing constitution document, making it official.

FBurkelwsd, 2023-07-25:
> I left a bunch of comments in the docs. I think we should assign someone to start from these documents and create a fresh update that we can iterate over. Once we're happy with the updated [...]

There are more now, still missing many. We have some easy updates with wording for new leadership structure but more tedious editing may need to occur in person. 

Final version could be finished within a week of in person

FBurkelwsd, 2023-08-07:
Pdf of Bylaws is in member resources of google drive.
Pdf of constitution is in member resources of google drive

FBurkelwsd, 2023-08-08:
Board should review the final drafts to be familiar with the process outlined. And to help update these documents.

I have reread the comments and have addressed all except for a possible in person meeting to rewrite the whole process.

geeknoid, 2023-09-05:
I think we're done, can we close this issue?

tcdoan, 2023-09-06:
@geeknoid  Sure. @Eshan-Feroz would you close this task?

closed 2023-09-06 by Eshan-Feroz