packofparts/frc2024 #59: Helping out new teams

BilledGarlic4, 2023-07-26:
A heavily seasoned and experienced FTC team (Velocity Raptors) is now transitioning to FRC. To get some good 'ol experience before competing in an actual season, they're building a Charged-Up Robot. They don't have a router yet but need a piece(s) cut (might have more pieces later on) until they get one. I'd really appreciate it if we could get the part cut out today. While this thread for now is for this team, we could expand it to help other rookie teams or teams with a lower budget. It's really good for impact, but more importantly, being able to share our resources that we're fortunate enough to have is really nice.

I'll post the CAM in a bit with details on setting up the file and the process the CAM will run so you guys can set it up. If we do choose to expand this effort to al teams who need resources i'll change the description later on.

BilledGarlic4, 2023-07-26:
Ok, so github won't let me upload a .gcode file, so i'll just throw it in Discord, in the CAD channel..

The files details/instructons are as follows:

- 22.125" x 1.5" x 0.25" (this is part size. stock, other than width, will be larger) 
- 1/4" polycarb stock (big piece)
- Starts with an 1/8" 2 flute bit and does 3 bores
- After that, changes to a 1/4" 2 flute bit 
- 1/4" bit does a pocketing pass on the inside of the part
- Then, 1/4" bit does a contour pass on the outside (with Tabs, make sure they don't disappear) to get the part done.
- chisel/bandsaw out part
- post process (file, debur, etc)
- Hand over to Ms. Dooley by end of today (Wednesday- 7/26/2023)

esjyoon, 2023-07-26:
I won't be here today (or next week), so does this piece have to be cut out today?

tcdoan, 2023-07-26:
> Ok, so github won't let me upload a .gcode file, so i'll just throw it in Discord, in the CAD channel..
>
>
>
> The files details/instructons are as follows:
>
>
>
> - 22.125" x 1.5" x 0.25" (this is part [...]

We are in the process to upgrade our GitHub account. Once the account moves above free standard level you should be able to attach your.gcode file to the task. In the meantime please upload to team 1294 google doc and update GitHub task to link to the google drive file.

closed 2023-08-26 by Eshan-Feroz