packofparts/frc2024 #72: Org Chart

FBurkelwsd, 2023-08-10:


FBurkelwsd, 2023-08-10:
Below is information on who is who for the 23-24 season

/

General
packofparts1294@gmail.com
Advisor & Head Mentor
fburke@lwsd.org
martin@taillefer.org
tcdoan@gmail.com
ben.stromseth@team1294.org
wygrangea@gmail.com

President 
ishita.jalla@team1294.org
Vice President
pradyu.kandala@team1294.org
Secretary
eshan.feroz@team1294.org
Project Manager
arvick.chandnani@team1294.org
Director of Business
shaan.kalgaonkar@team1294.org
Director of Public Relations
arisha.gupta@team1294.org
Director of Mechanical
ethan.yoon@team1294.org
Director of Electrical
aditya.shah@team1294.org
Director of Programming
teja.koripella@team1294.org
Director of Design/CAD
sanchit.arora@team1294.org
Director of Materials
maahi.modi@team1294.org

ArvickC, 2023-09-09:
@Powerlax Do you have any updates on this issue?

Powerlax, 2023-09-09:
Currently working on gathering and adding pictures, as well as integrating with website

ArvickC, 2023-11-29:
org chart has been updated and can be found [here](https://www.canva.com/design/DAFuYdqW0-M/UrhrEgX_XNMTHTevyD3zbg/edit?utm_content=DAFuYdqW0-M&utm_campaign=designshare&utm_medium=link2&utm_source=sharebutton)

closed 2023-11-29 by ArvickC