packofparts/frc2024 #73: Replace Swerve Modules

Ants1the1, 2023-08-10:
Convert the current drive base of Riptide to use the MK4 Swerve modules. Keep using the Sparks that are already on the drive base.

aditya0shah, 2023-08-10:
@TechTaco1 Work with programming to label sparks with their CAN id after wiring them to mk4.

TechTaco1, 2023-08-11:
After today, one module is fully wired and routed, complications in wiring for front right module as there is no space for sparks, back left module as ther is an old neo

Ants1the1, 2023-08-12:
Mechanically I believe the modules are completely finished. They are mounted, assembled and have loctite on the bolts that need them.
We might change half bolts that attach them to the frame for longer Phillips head 10-32 bolts to allow us to attach our bumpers.[image]

closed 2023-08-26 by Eshan-Feroz