packofparts/frc2024 #77: Drive team selection

FBurkelwsd, 2023-08-11:


FBurkelwsd, 2023-08-11:
We need to get dates, and a plan for how.

Then schedule with Project manager and team for robot use

FBurkelwsd, 2023-09-16:
Drive Team discussion notes
Goal is to selection earlier than last year and Use the off season to test potential drive teams
Please make it unobtrusive as possible for drive team selections because Riptide needs work
Look for current juniors or lower for tryouts.
Second drive team didn’t work great, but backups for each position are good
Technicians – 4 were a lot, rotating them out might be a good idea 
Somebody on the drive team from each subteam would be good
Drive team should not be fixing the robot. We should have a pit crew for that with at least one from each subteam on the pit crew: mech, elec, prog.
We chose people who were top of their field, should we have chosen the apprentices? If we do have the top they need to designate an alternate to help.
Off-seasons aren’t the time to show off and be the best, it’s prep for the on season.
Work on games that make the team better, and communication building exercises.
Get coach input after the off-season comp as well as the driver teams as well. Those people will have the boots on the ground input on what’s going on with each other.
We agree off-season should be the proving ground for the drive teams.
We agree the feedback from the parts of the team will be important for making a final decision.
Send a survey to ask who would like to be a part of Girls Gen drive team.
Send a survey to ask who wants to try out for regular drive team. (be sure to specify roles available)
How many people are we going to try out for drive team and what are we going to do to train them before driving.
We suggest narrowing it down to no more than 3 driver/operator pairs and coaches. 
How do we determine those pairings before the event?
Akshai to reach out to Girls who Code coach about getting some of them to drive – this is a good recruiting opportunity.
How involved do we want incoming freshman to be? Will they be able to be a part of off-season drive teams? Even if the first meeting is the 20th?
How are we selecting coaches and drivers?
Should we enforce required practice hours on the bot – how will we tally that?
How much time will you give people in pre-season to practice before choosing the final drive team?
What’s our preliminary barometer for choosing the coaches?
•	Interview (what do we want in our coaches?)
•	Years on the team (years of experience)
•	Use training games to find out who can do what. (robot-driver-coach communication game. Defuse the bomb. Etc)
3 stages of picking the drive team:
•	Coaches – narrow them down first (6 down to 3 eg)
•	Driver pairs – try them out with the coaches
•	Off-season comps where we mix and match and determine results.
•	Pick the pit crew, should there be a pit boss/manager? What would that person look like? 
•	Don’t forget the safety captain – add to board agenda

Make sure people have the time commitment to be at all comps, and get to practices
Get parents to agree to allowing kids to come to comps, make sure we have a timeline for comps and such to present to parents. This will avoid last minute subs due to vacation plans.

Powerlax, 2023-09-27:
Rough plan made with dates and goals for each day on the google doc

FBurkelwsd, 2023-09-27:
https://docs.google.com/document/d/1XaQ5TlC5pa67t_1UIsdoPcvLrLjVUwWE_0m7coxWZiI/edit

Awesome lets keep a link here to make documenting easier

FBurkelwsd, 2023-09-27:
Okay can we talk through this calendar later today?

FBurkelwsd, 2023-09-27:
We are going to have to reserve the robot for specific days or times during our regular meeting

closed 2023-10-30 by ArvickC