packofparts/frc2024 #87: Onshape Sponser

FBurkelwsd, 2023-09-06:
[desktop.pdf](https://github.com/FRC-1294/frc2024/files/12541381/desktop.pdf)
the pdf is a screen shot of the full application.

Questions: we need to collaborate on 
Diversity and Inclusion
PTC and FIRST share a commitment to diversity and inclusion so that all young people have an opportunity to benefit from the exceptional learning experiences these programs offer.

A recent study by Brandeis University found that participants in FIRST gain interest and skills in STEM disciplines from their experiences. It also found that female students, African American/Black students, Hispanic students, and students on free/reduced lunch (living in or near poverty) are not equally represented in our leagues and on our teams.

At PTC we believe it is essential STEM education programs, like FIRST, are accessible to everyone in order to develop future technology leaders who reflect the gender, racial, and ethnic background of the population.
Please describe the makeup of your team and how you promote a culture of inclusivity*
How do you recruit members and create a sustainable team?
Describe how your team recruits team members. Are there specific efforts made to attract and retain students from underrepresented groups? Also, describe how you ensure the team is sustainable, whether it be through recruitment, sponsorships, or other means.
What does your team do for outreach?

FBurkelwsd, 2023-09-15:
[onshape draft 1 grant.pdf](https://github.com/FRC-1294/frc2024/files/12629152/onshape.draft.1.grant.pdf)

SanchitArora44, 2023-09-16:
[image]

some things to review:

"Please select the software tools your team currently uses."
* OnShape isn't selected in the highlighted area

"What Other Software tools does your team use?"
* We could also mention VS Code and PhotonVision in regard to the programming tools
* The line "We use Autodesk Fusion 360 to interface with our manufacturing machinery" isn't exactly accurate. Fusion 360 doesn't interface with the CNC, it simply writes G-code for it. Instead, the line could be changed to "We use Autodesk Fusion 360 to do CAM and create G-code and use Mach-3 to interface with our CNC machines."
* "Tormach tool interface" can be changed to "Path Pilot", Tormach's official name for their software
* I might be wrong about this one so I'd appreciate if someone could correct me on this one if I'm wrong, but the question says, "What other software's does your team use", so mentioning OnShape doesn't help answer the question in any way
* We don't use "Rev Online", and I don't believe any such thing exists. I think you might be referring to "REV Hardware Client."
* Not a necessity, but mentioning which specific slicers we use might help? idk

"PTC software past usage and future plans"
* Some other reasons we like OnShape other than those mentioned: featurescripts, saved on the cloud, auto saves if something crashes, data management made easy (versions, branching, merging), multiple features not offered in other studios, variable studios and variables, intuitive and easy for students to learn, the OnShape learning center with numerous courses for all levels of CAD'ers (OnShape learning center point is repeated later on so can be left out)

"How do team members learn CAD"
* Below the answer box they mention linking any external/self-created resources we use, I'm in the process of creating PowerPoints for CAD lessons, so we can use those

"How do you support other teams?"
* The summer camps point might be better suited for the previous question as it talks about how we train _our_ members, and not how we help support other teams
* We can also talk about how we've helped start two FTC teams, and manufactured parts for multiple teams in the nearby area, including rookie teams

"Please describe the makeup of your team...." & "How do you recruit members..." & "What does your team do for outreach"
* This is really well done imo, but if exec has more to add it definitely wouldn't hurt (i can't see if we've mentioned stem kits or the summer camps in the outreach question, but that might help a little?)

"Would your team be interested in the following opportunities?"
* I'd love to do some of these, but since exec will probably be behind the video production/social media, it'll be their decision in the end

FBurkelwsd, 2023-09-16:
[onshape draft 2.pdf](https://github.com/FRC-1294/frc2024/files/12639707/onshape.draft.2.pdf)
ignore the top information and the non clicked boxes. they are actually clicked.

jjalla, 2023-09-17:
How do you recruit members and create a sustainable team?

Recruiting members has always been an essential part of our team. We start with the broad areas. Posting fliers around the school, having our club mentioned in the morning announcements, having teachers plug our club in their classes, etc. However, recently we started going more in depth and trying to cover all our bases. We aim for our team to be more inclusive and represent how the world looks like. We started targeting underrepresented groups specifically and talking to them. For example, theres a Girls Who Code club. A club that's very STEM oriented and also made up of women. By specifically targeting this club we aim to recruit more women. Regarding sustainability, one of our outreach events, the recycling initiative mentioned in the question below, specifically targets reducing more waste within the FIRST community and promotes reusability.

jjalla, 2023-09-17:
For the outreach question there seems to be a lot of information that might not be necessary because we pulled from the Chairmen's essay. Let's try to consolidate. Is there a way @FBurkelwsd can send the whole answer here so we can edit it?

jjalla, 2023-09-17:
Please describe the makeup of your team and how you promote a culture of inclusivity:

Our team is made up of made up of students from all over the school district, promoting different perspectives. Our board is made up of the President, Vice President, Secretary, Director of PR, Director of Business, Project Manager, Director of Electronics, Director of Mechanics, Director of CAD/Fabrication and Director of Programming. These leadership positions are comprised of students of color and different genders. Our president is a woman, and has made championing different perspectives and creating a culture that welcomes all opinions equally a priority for the team. We strive to have a team with members and mentors that encompasses different ideas and various viewpoints.

FBurkelwsd, 2023-09-17:
> For the outreach question there seems to be a lot of information that might not be necessary because we pulled from the Chairmen's essay. Let's try to consolidate. Is there a way @FBurkelwsd can send the whole answer here so we can edit it?

 The Motor

The journey of 1294 Pack of Parts is a story of growth and evolvement. Like gears in a machine, each aspect of our team is interlocked and built upon the other to create a well-oiled system that creates a lasting impact on the team and community. The spirit that once sparked a fire in 2004 has only grown, fueling our members to continue to innovate and tackle new challenges. Each gear in our team turns with the others, creating a machine, although not flawless, has the capability of achieving great things. As every machine needs a motor to start it, the passion within our team is the driving force that propels us forward and drives us to share our experiences with others.

The Organization Gear 
Just as the starting gear is essential to move the entire system, the organizational structure of 1294 is crucial in bringing the team's diverse subteams together and ensuring that they all work effectively toward a common goal.  
With about 90 members annually, we divide technical and nontechnical work into five subteams, each group varying in size from 10 to 35 students. The build/CAD subteam is responsible for designing, prototyping, and manufacturing our robot drive bases and mechanisms. Electrical subteam members work alongside build/CAD to wire and test dynamic components. On the other side of the shop, programming diligently writes and tests robot code in smaller, project-specific tasks. Our PR/Business subteam manages our social media platform, sponsorships, finances, and increases team spirit. Each subteam is guided by a director who oversees a group of leads. They work to organize and pilot the subteam, ensuring continuity and growth in the coming years. 
To ensure that organization remains the driving force behind our success, in fall of 2020, we launched a series of online training programs designed by our veteran subteam leads. This introduces new members to standard FRC design, build, and programming practices. Since then, the training programs have served as a stepping stone for new members into the world of FIRST. For example in the 2020-21 season, over 60% of our members possessed at least a beginner-level proficiency in CAD. The addition of technical documentation and in-depth training for members has allowed us to experiment with new manufacturing techniques, such as CNC cutting and 3D printing.
Moreover, we have industrialized our robot building process. While we once scoured the scrap bin for intake mechanism material, we now develop robust plans to source necessary materials and machine parts with confidence. Organization acts as the backbone that holds everything together and ensures that each piece is operating, allowing the team to grow and evolve.

The Outreach Gear 

Working alongside organization, outreach is another vital gear which keeps us moving forward. As we cleaned our shop space in preparation for the 2022 Build Season, we uncovered bins full of parts we no longer used. We thought; “What if we could give these parts to teams that need them?” From this question, the Pack up your Parts Recycling Initiative was born, a pop-up FRC donation center operating out of our team’s shop space at Eastlake HS. Teams bring their unused parts and pick up available parts from other teams to create a sustainable cycle of robot parts. We created a public spreadsheet with a list of available parts so teams can reserve what they need. This year, we set aside a designated time each meeting to allow for local teams to come and donate to our 2nd Annual Recycling Initiative. 
As one of the largest gears, outreach plays a significant role in strengthening our system. Pack of Parts is very fortunate to be sponsored by Microsoft, Aerojet Rocketdyne, Boeing, and more. Our sponsorships begin with reaching out to local companies who are potentially interested in funding our team. Then, they review our tiered sponsorship system to determine how much they would like to invest and what benefits they receive. Mentors who are employed at specific companies match their volunteer hours for team donations. Additionally, to convey our gratitude towards sponsors we send frequent thank you cards signed by our members and updates regarding our season.
With funding from companies such as Aerojet RocketDyne that support our mission to educate young passionate students, Pack of Parts spent the summer designing, creating, and manufacturing STEM kits. We designed a total of 5 stem kits: a crazy catapult, tensegrity table, scissor lift, disc launcher, and planet explorer. These kits were aimed to provide hands-on learning opportunities for students interested in STEM. Throughout the year, we focused on the distribution and manufacturing of the Crazy Catapult STEM kit at various community events such as farmers markets, holiday bazaars, and at library functions, reaching about 200 people over the last year. At these meets, we collected feedback from participants to ensure that we were improving our designs. The kits were targeted toward elementary and middle schoolers which taught them the core engineering principles of the energy flow of a catapult using our designed parts. Additionally, to expand our stem kits beyond our immediate community we met with government off [... cut ...]
As Pack of Parts, we are more than just a robotics team. Each of our members bring unique skills, perspectives, and experiences, and together, we form a cohesive and powerful unit, fueled by passion and made stronger through the different parts. We are like the pieces of a puzzle that fit together to create something beautiful and meaningful. The end result is a group that is stronger and more capable than any individual member could be on their own. Whether it's through volunteering at FIRST events or making a difference in our community, we are Pack of Parts, dedicated to making a positive impact in the world.

FBurkelwsd, 2023-09-17:
[onshape draft 3.pdf](https://github.com/FRC-1294/frc2024/files/12643123/onshape.draft.3.pdf)
will check back again today to submit

shaankal, 2023-09-17:
For the “What does your team do for outreach” Question: Our team has a variety of outreach events, that allow for FIRST to be spread throughout our nearby community. This past summer, we hosted 3 STEM summer camps, ranging from grades 6-9. In our camps, our team members taught the younger generation all there needs to be learned about FIRST Robotics. We taught CAD, programming, engineering, and other necessary skills to be successful in FIRST. Over 70 students signed up for our camps this summer, and we received great feedback.Outside of our STEM camps and workshops that  expose children to engineering from a young age, our team hosts variety of initiatives, such as Food FIRST. Our Food FIRST Initiative began back in March 2020, as a single team fundraiser. Today, this fundraiser has reached 24 FIRST teams in 6 states raising $4,000 in donations over two months. Alongside Food FIRST, our team created our own Recycling Initiative which has become a yearly tradition. It's a pop-up thrift store with the goal of reducing the amount of parts that go to waste within FRC teams and encourage sustainability. FRC teams from all over the community come to our shop and drop off parts that they don't need, and then are encouraged to pack up parts from our shop that other teams have dropped off. This way parts are cycled throughout the community and the waste that our community produces is decreased significantly. Another initiative that has grown into its own brand and product are STEM kits. These are easy to build kits produced in our shops that have various components and instructions so students can engineer a project on their own. We are aiming to bring these kits into classrooms in Title I schools to help underprivileged kids experience the joy of engineering.

FBurkelwsd, 2023-09-18:
Submitted! We are making significant progress in our collaboration!

closed 2023-09-18 by SanchitArora44