packofparts/frc2024 #88: Aerojet Rocketdyne

FBurkelwsd, 2023-09-06:
[image]

Need help drafting the responses for the attached images questions. 

I created an account to finish the application.

FBurkelwsd, 2023-09-10:
We strive to be a student led team with mentors who give guidance and support as needed. We share our space with engineering, robotics, theatre technology, outdoor adventures, and material science classes which gives our team a complex experience. 

We embody our passions for all things STEAM related. Organizing and holding summer camps for middle school for programming, build and electrical. Our students created and are developing stem kits which empower any student to learn the engineering/design process through hands on student created activities. 

We are always learning from feedback and prior experiences. We  have our first season of swerve drive base completed, and are now able to maintain and optimize multiple functional drive bases for all members to implement innovative solutions. 

Our student board is working together with multiple digital resources. They are striving to make all sub teams work and ideas available for all our upcoming members to be able to contribute and understand how our team is moving forward.

shaankal, 2023-09-16:
I can work on these questions. Can you attach a link to the website?

FBurkelwsd, 2023-09-16:
> I can work on these questions. Can you attach a link to the website?

IDK? This one originated in an email to me.
https://www.l3harris.com/company/aerojet-rocketdyne

FBurkelwsd, 2023-10-09:
Great news! We received the Aerojet Rocketdyne grant for Eastlake Robotics program in the amount of $1,000! I will pay this out to the district today, and you should be receiving the funding by the end of the month.

FBurkelwsd, 2023-10-11:
COMMUNITY GRANTS EVALUATION FORM
Like all non-profit institutions, we regularly report to our donors how effective our funded programs further our goals as an educational foundation. We are dependent upon our grant recipients to evaluate the success of their community grants and to share with their community that LWSF has provided the funding. Our ability to obtain future grant funding is directly impacted by the quality and thoroughness of the grant evaluations received as well as the public knowing the funding is from LWSF.
Please submit this form via email no later than June 1, 2024. If your grant program will run over this deadline, please email [info@lwsf.org](mailto:info@lwsf.org) by May 15th to request an extension. Failure to submit a timely evaluation may impact future grant funding.
PROJECT NAME:

PROJECT LEADER(S) NAME(S):

LEAD APPLICANT’S NAME:

LEAD APPLICANT’S EMAIL ADDRESS:

1.	PARTICIPANTS: How many participants did you serve?

2.	DEMOGRAPHICS: What population did you serve? (Please describe where your participants live, how old they are, and any other relevant and known demographics.)

3.	PROBLEM: What problem(s) did your program attempt to solve or mitigate?

4.	ACTION: What specific actions did your program take to help solve the problem(s)? (Please indicate the type and specific number of events or actions undertaken.)

5.	EVALUATION: What specific measurements or outcomes did you tally?

	
6.	GOAL: Did your program accomplish the specific, measurable improvement(s) that it set out to achieve?

7.	DISCUSSION: 
a.	What impact did your work have? Did anything change as a result of your work?

b.	Did any aspect of your work not go as planned? Would you implement changes to your procedure or other operations in the coming year?

c.	How did you acknowledge LWSF’s support?

d.	Was there a memorable moment of transformation or elucidation that made you think your project was working? Could you describe a positive change observed over time in an individual or within the group? Please anonymize your anecdote; we would like to share these stories with our donors.

Please take photographs showing your project in action and email them to us at [info@lwsf.org](mailto:info@lwsf.org). A photo release form is required for anyone identifiable in an image. You will receive a copy of the photo release form with this evaluation, but please email [info@lwsf.org](mailto:info@lwsf.org) if you need the photo release form to be resent.

FBurkelwsd, 2023-10-11:
[Community Grants Acceptance Form 2023-2024 Eastlake HS Robotics Program.docx](https://github.com/FRC-1294/frc2024/files/12873221/Community.Grants.Acceptance.Form.2023-2024.Eastlake.HS.Robotics.Program.docx)

FBurkelwsd, 2023-10-11:
https://form.jotform.com/220808248659162 for photo release when we submit photos

ArvickC, 2023-11-28:
Aerojet Rocketdyne agreed to sponsor $1000 towards our team

closed 2023-11-28 by ArvickC