packofparts/frc2024 #102: BAE grant

FBurkelwsd, 2023-09-15:
Due sept 22
[bae grant outline.pdf](https://github.com/FRC-1294/frc2024/files/12624038/bae.grant.outline.pdf)
we have some more questions at the bottom. I will be able to pull a lot from our impact award essay, and our rev grant
https://docs.google.com/document/d/1jqH_S2aSn4AUnkwKTzJKJwedB7kwDTKbwDUHRGk0ngU/edit

shaankal, 2023-09-16:
I can do the last 2 questions on it. I will need help with basic info (Ex: if we have anyone working for BAE, free lunch percentage , demographics)

FBurkelwsd, 2023-09-16:
I will be able to fill out the numbers

FBurkelwsd, 2023-09-20:
After rereading this application, I may be able to fill out the whole thing. will post a copy here later today. Please review or comment before friday

FBurkelwsd, 2023-09-20:
Farmers Market STEM outreach and Robt showcase at Sammamish city hall
We host summer camps for both engineering and programming for middle school students and incoming freshman. 
We designed and are marketing STEM kits for middle and elementary school students to explore engineering concepts at home.
We have been holding a recycling initiative for the past years, hoping to reduce the number of robotic components that end up in the landfill.
Our Food FIRST Initiative began back in March 2020, as a single team fundraiser. Today, this fundraiser has reached 24 FIRST teams in 6 states.

[image]
[image]

FBurkelwsd, 2023-09-20:
do we have members who are also apart of girls who code

FBurkelwsd, 2023-09-21:
Awards we have won, 
2021 excellence in Engineering
2019 PNW Glacier Peak Winner
2022 Judges Award Bonney Lake
2022 Judges Award Sammamish HS
2017 Quality award Auburn
2017 PNW Glacier Peak Finalist
2019 PNW Glacier Peak Winner
2023 Excellence in Engineering Award

jjalla, 2023-09-21:
> do we have members who are also apart of girls who code

yes

jjalla, 2023-09-21:
List one event per the following format. Year, Event name, Participation, or host. A brief one-sentence description of the event.

2023, Programming Camp, Host. We hosted a programming camp for grades 7-9 and taught them the basics of Java and WPIlib.
2023, Engineering Camp, Host. We hosted an engineering camp for grades 8-9 and taught them the principles of engineering how to use tools, and how to CAD.
2023, Engineering Camp, Host. We hosted an engineering camp for grades 6-7 and taught them the basics of engineering, how to use tools, and how to CAD.
2023, Sammamish Farmer's Market, Participant. We participated in our community's farmer's market and engaged with our community telling them about robotics and our club. 
2023, Sammamish Farmer's Market, Participant. We participated in our community's farmer's market and engaged with our community telling them about robotics and our club. 
2023, Pack Of  Parts Recycling Initiative, Host. We ran a pop-up "thrift shop" for FRC parts for teams in our communities. They could drop off parts they don't use and pick up parts that other teams dropped off. 
2023, Robot Reveal Food Drive, Host. We used our robot reveal as a food drive opportunity and donated food to homeless shelters.
2023, Mead STEM night, Participant. We went to an elementary school and exposed children to STEM and what we do as a club.
2023, STEM Kits workshop, Host. We ran a workshop for children to use our STEM Kits to test our product and see how much children gain from it.
2023, Advocating STEM education access with King County, Participant. We advocated for STEM education in our county. 
2023,  FIRST WA Youth DEI Committee, Participant. We participated in the Youth committee to advocate STEM education and diversity.
2022, Engineering Workshops, Host. 
2022, Programming Workshops, Host. We hosted a programming camp over Zoom and taught the fundamentals of Java.
2022, Electrical Workshop, Host. We hosted a 5-hour workshop at our library regarding circuitry and electricity for elementary school. 
2022, Issaquah Salmon Days, Participant. We participated in a Washington festival and talked about our club and STEM. 
2022, Sammamish Farmer's Market, Participant. We participated in our community's farmer's market and engaged with our community telling them about robotics and our club. 
2022, Sammamish Farmer's Market, Participant. We participated in our community's farmer's market and engaged with our community telling them about robotics and our club.

FBurkelwsd, 2023-09-22:
[bae grant.pdf](https://github.com/FRC-1294/frc2024/files/12704532/bae.grant.pdf)
submitted 

Thank you for the help

closed 2023-09-25 by ArvickC

FBurkelwsd, 2023-10-30:
Thank you for applying for a FIRST® team grant from BAE Systems. We wish we could respond favorably to all teams who request a grant, however, that is not possible given the volume of applications we receive.
 
We regret that we will not be able to support your team this year.

We encourage you to apply for a grant during next years’ application process and wish you the best of luck in your upcoming season!
 
If you have specific questions, please reach out to us at [bae.firstrobotics@baesystems.com](mailto:bae.firstrobotics@baesystems.com).