packofparts/frc2024 #103: WCP grant

FBurkelwsd, 2023-09-16:
https://www.chiefdelphi.com/t/wcp-2023-24-product-release-updates/438929/56
Deadline is October 31st
[WCP grant.pdf](https://github.com/FRC-1294/frc2024/files/12639855/WCP.grant.pdf)

FBurkelwsd, 2023-10-11:
How would WCP grant help our team?

shaankal, 2023-10-19:
Need to clarify with Sanchit on the type of sponsorship we want. Other than that, the responses should be done by tonight/tmrw morning...

shaankal, 2023-10-19:
Might need support on this question : How would a WCP parts sponsorship help your team?

FBurkelwsd, 2023-10-20:
[image]
what they sell. we did get kraken from this distributer

shaankal, 2023-10-31:
How would a WCP parts sponsorship help your team?

How active is your team during the off season? Briefly describe any activities projects your team does in the off season.
Our team hosts a plethora of activities throughout our offseason. This past summer, we hosted our very first engineering and programming summer camps. These summer camps, meant for grades 6-9, was a way for our team to help the next generation of STEM innovators. Within this camp, kids learned about the design process, how to use CAD programs, as well as technical products such as drills and saws. We also taught incoming members how to program for the upcoming season, which includes WPILIB and Java. 
Does your team help/mentor any FTC teams in your area or remotely?   If yes please provide more information (teams, etc..)
We recently begun to mentor team 23368 Nova Robotics for this upcoming FTC season. We sponsored them with $1000, in order to jumpstart their season. Throughout their season, we will have our workshop open for any reason that they desire, and communications will always be between us. As an experienced team, our sponsorship and mentorship to Nova can guarantee further success in their season. 

Does your team help/mentor other FRC teams in your area or remotely? If yes please provide more information (teams, etc..).
Our team has our recycling initiative, a way for expand and create a community. Teams bring their unused parts and pick up available parts from other teams to create a sustainable cycle of robot parts. We created a public spreadsheet with a list of available parts so teams can reserve what they need. Through this program, it has been a way for our team to help others throughout our build season, which is something our team strives for. 
How would a WCP parts sponsorship help your team?
A WCP parts sponsorship would greatly benefit our team. Our team would be able to receive general parts, that would greatly improve our chances to compete well in the competitions later in the season. Being able to receive parts from WCP allows for us to strengthen relations further, ensuring future connections regarding grants. In addition, being able to test products at the beginning of build season because it would allow for us to have the ability to use all of the up to date products from WCP. Overall, receiving parts such as Krakens, and other valuable parts is perfect for our team, and receiving a sponsorship would ensure this.

FBurkelwsd, 2023-11-01:
Submitted today

closed 2023-11-02 by shaankal