packofparts/frc2024 #107: Lake Washington School Foundation Grant

FBurkelwsd, 2023-09-22:
this one isn't available until 2024.

This one would be a good opportunity t
[LWSF grant community.pdf](https://github.com/FRC-1294/frc2024/files/12704538/LWSF.grant.community.pdf)
o have a community project such as Stem kits and or summer camps or new project.

FBurkelwsd, 2023-10-11:
for this grant: when we apply.

will have to use the following form to document our year.

COMMUNITY GRANTS EVALUATION FORM
Like all non-profit institutions, we regularly report to our donors how effective our funded programs further our goals as an educational foundation. We are dependent upon our grant recipients to evaluate the success of their community grants and to share with their community that LWSF has provided the funding. Our ability to obtain future grant funding is directly impacted by the quality and thoroughness of the grant evaluations received as well as the public knowing the funding is from LWSF.
Please submit this form via email no later than June 1, 2024. If your grant program will run over this deadline, please email [info@lwsf.org](mailto:info@lwsf.org) by May 15th to request an extension. Failure to submit a timely evaluation may impact future grant funding.
PROJECT NAME:

PROJECT LEADER(S) NAME(S):

LEAD APPLICANT’S NAME:

LEAD APPLICANT’S EMAIL ADDRESS:

1.	PARTICIPANTS: How many participants did you serve?

2.	DEMOGRAPHICS: What population did you serve? (Please describe where your participants live, how old they are, and any other relevant and known demographics.)

3.	PROBLEM: What problem(s) did your program attempt to solve or mitigate?

4.	ACTION: What specific actions did your program take to help solve the problem(s)? (Please indicate the type and specific number of events or actions undertaken.)

5.	EVALUATION: What specific measurements or outcomes did you tally?

	
6.	GOAL: Did your program accomplish the specific, measurable improvement(s) that it set out to achieve?

7.	DISCUSSION: 
a.	What impact did your work have? Did anything change as a result of your work?

b.	Did any aspect of your work not go as planned? Would you implement changes to your procedure or other operations in the coming year?

c.	How did you acknowledge LWSF’s support?

d.	Was there a memorable moment of transformation or elucidation that made you think your project was working? Could you describe a positive change observed over time in an individual or within the group? Please anonymize your anecdote; we would like to share these stories with our donors.

Please take photographs showing your project in action and email them to us at [info@lwsf.org](mailto:info@lwsf.org). A photo release form is required for anyone identifiable in an image. You will receive a copy of the photo release form with this evaluation, but please email [info@lwsf.org](mailto:info@lwsf.org) if you need the photo release form to be resent.

FBurkelwsd, 2023-10-11:
https://forms.office.com/pages/responsepage.aspx?id=GpWzqH5GvUaXzy8GPWVrAR1TQ0T_2ttCtIn0NAqLsGdUOEtYNkhTU1FaMDJYU0dHRjhWMU4yUVE4RC4u link to application for additional funding

FBurkelwsd, 2023-10-11:
https://form.jotform.com/220808248659162 for photo release when we submit photos

ArvickC, 2024-01-27:
Responses are on the drive

closed 2024-01-27 by ArvickC

FBurkelwsd, 2024-02-20:
> Responses are on the drive

no reason to not put the direct link