packofparts/frc2024 #108: Ospi grant

FBurkelwsd, 2023-09-22:
[firstwa_frc ospi grant.pdf](https://github.com/FRC-1294/frc2024/files/12704796/firstwa_frc.ospi.grant.pdf)
This one is due by September 29th

I have been able to take some from past applications.

FBurkelwsd, 2023-09-22:
[image]
[image]
[image]
[image]

SanchitArora44, 2023-09-23:
7 days left!

FBurkelwsd, 2023-09-26:
Student Recruitment Plan ?
How do you recruit members and create a sustainable team?
Recruiting members has always been an essential part of our team. We start with the broad areas. Posting fliers around the school, having our club mentioned in the morning announcements, having teachers plug our club in their classes, etc. However, recently we started going more in depth and trying to cover all our bases. We aim for our team to be more inclusive and represent how the world looks like. We started targeting underrepresented groups specifically and talking to them. For example, there's a Girls Who Code club. A club that's very STEM oriented and also made up of women. By specifically targeting this club we aim to recruit more women. Regarding sustainability, one of our outreach events, the recycling initiative mentioned in the question below, specifically targets reducing more waste within the FIRST community and promotes reusability.

FBurkelwsd, 2023-09-26:
[image]
@shaankal @pk2107 Can we get a draft for this answer before thursday. Feel free to use something you have already created

FBurkelwsd, 2023-09-26:
For the what does your team do to help teams in your area from shaan
For the “What does your team do for outreach” Question: Our team has a variety of outreach events, that allow for FIRST to be spread throughout our nearby community. This past summer, we hosted 3 STEM summer camps, ranging from grades 6-9. In our camps, our team members taught the younger generation all there needs to be learned about FIRST Robotics. We taught CAD, programming, engineering, and other necessary skills to be successful in FIRST. Over 70 students signed up for our camps this summer, and we received great feedback.Outside of our STEM camps and workshops that expose children to engineering from a young age, our team hosts variety of initiatives, such as Food FIRST. Our Food FIRST Initiative began back in March 2020, as a single team fundraiser. Today, this fundraiser has reached 24 FIRST teams in 6 states raising $4,000 in donations over two months. Alongside Food FIRST, our team created our own Recycling Initiative which has become a yearly tradition. It's a pop-up thrift store with the goal of reducing the amount of parts that go to waste within FRC teams and encourage sustainability. FRC teams from all over the community come to our shop and drop off parts that they don't need, and then are encouraged to pack up parts from our shop that other teams have dropped off. This way parts are cycled throughout the community and the waste that our community produces is decreased significantly. Another initiative that has grown into its own brand and product are STEM kits. These are easy to build kits produced in our shops that have various components and instructions so students can engineer a project on their own. We are aiming to bring these kits into classrooms in Title I schools to help underprivileged kids experience the joy of engineering.

shaankal, 2023-09-27:
For the additional funds question, i can get that done by Wed night...

shaankal, 2023-09-28:
What is your plan for raising additional funds to support your team.
	Our team has a variety of techniques in order to raise funds. Throughout the year, our team hosts a variety of fundraisers, such as MOD Pizza and Chipotle. We market these events throughout our school and the surrounding community. Through this, we gain overarching support from those around us, raising additional funds outside of sponsorships and grants. Another technique is through our outreach, specifically Farmers Markets. We visit the the Sammamish Farmers Market at least 3 times a year, at these events, we display our robot and the overall FIRST community. We receive donations during these events, gaining additional funds for our team.

How do you empower students for new things
	This year, we introduced our round-robin for our first meeting. During this event, new and returning members were split up into 6 groups, representing the different subteams, and were traveling to different stations, where the directors were waiting. At these stations, the directors explained their subteam and why you should join. This tactic gave us spread results, with new and returning members showing interest in different subteams such as Business and Public Relations. We are also introducing the idea of rationing the subteams, which will ensure no subteam has a larger number of people than another. This way, members are exposed to new subteams, which increases the chance of finding a new passion on the team.

FBurkelwsd, 2023-09-29:
Our team is comprised of many different sub teams (mechanical, electrical, programming, business, public relations, materials, and executive) to give all members an opportunity to contribute based on interest. We introduce our round-robin for our first meeting where new and returning members were split up into 6 groups, and we immersed in each teams roles with past and future projects.  In these groups, the directors explained their subteam and why you should join. This tactic gave us spread results, with new and returning members showing interest in different subteams such as Business and Public Relations. We are also introducing the idea of rationing the subteams, which will ensure no subteam has a larger number of people than another. This way, members are exposed to new subteams, which increases the chance of finding a new passion on the team. As the year goes along members have conflicts in schedule, so many students get the opportunity to be apart of and create multiple sub team projects.

FBurkelwsd, 2023-09-29:
Submitted this

FBurkelwsd, 2023-10-12:
[image]