packofparts/frc2024 #115: Get designs reviewed by mentors

ArvickC, 2023-09-25:


FBurkelwsd, 2023-09-25:
[image]
We intended for the work bench to be incoporated into the milwaukee storage. Remember we need to fold the two 8020 modules and lock together to transport, as tipping could be an issue. Also need a way to secure the 8020 pit when set up to prevent it from falling over. 
[image]
Use the 1727 pit as inspiration of what our 8020 design could be. with each drawer area being a place for a milwaukee bin to be.

SanchitArora44, 2023-09-25:
Right now, the agreed upon plan is to make the two storage units out of 80-20 (they have more than enough storage for Milwaukee packout's by a large margin), and while they are being built, we CAD the workbenches. If time is left AND we are happy with the way the storage units turned out, then we move on with building the workbenches. If there is not enough time left, or we are not happy with the way the storage turns out, we will buy the husky workbenches and either use them as is or modify them slightly. The husky workbenches have plenty of space for power tools within the larger drawers. 

From all the members I've talked to, we all agree that simply buying the husky workbenches will be much more time, energy, and cost efficient, than building the workbenches ourselves this season. Personally, i'd like to build the storage areas, buy the husky workbenches, and incorporate these workbenches into another pit structure in the future. There are so so many projects the team is trying to do, that trying to make our own workbench structure is, in my opinion, not feasible.

FBurkelwsd, 2023-09-25:
https://catalogs.8020.net/80-20-Catalog-24/i/

We selected 8020 due to the customizable nature. And we have a very relevant design and build projects,
[image]
I notice we are missing wheels and hinges. also there are sliders for our own drawers.
[image]
[image]

FBurkelwsd, 2023-09-27:
Please check this:
https://github.com/FRC-1294/frc2024/issues/75

FBurkelwsd, 2023-10-11:
parts to incorporate.
https://www.mcmaster.com/47065T356 hinges for folding the storage together and locking in place
https://www.mcmaster.com/47065T357/
https://www.mcmaster.com/47065T345/
https://www.mcmaster.com/47065T356/

https://cad.onshape.com/documents/3403834b07e4f3fc1af36460/w/34ad64f2a03a9e5897971e8c/e/fbe575708f3503b621353935

https://8020.net/2703.html
https://8020.net/65-2323.html
are possible casters for the bottom

FBurkelwsd, 2023-10-11:
Why buy a husky bin when we can make a custom push cart and use an extra shop table top as the work bench

SanchitArora44, 2023-10-11:
Because overall it is much cheaper, time efficient, sturdy, and does exactly what we need it to do. Furthermore, in later seasons if we want to make our own workbench, you can very easily integrate this. This is one of the reasons i've brought up the issue of prioritizing projects. The storage alone has been in the design phase for 3 months without anything to show in terms of prototyping or mfg, so making our own workbench will only add to the time commitment. 
All subteams have many other projects that are more important to focus on than a workbench that we can buy. 
I've talked to multiple other members, and they all agree that we should build the storage but buy the workbenches due to both the time saved by doing so, as well as the guarantee of it being well made.

FBurkelwsd, 2023-10-11:
We haven't focused on this project well enough. We are to scattered with the loss of substantial members last year. Once we get our new members safety training and through subteam practice we will have plenty of help to manufacture. one reason why extra off season projects is not needed.

SanchitArora44, 2023-10-18:
Considering that the current roll out carts in our shop are roughly 6.5-7 feet tall, and 20" wide, and are yet to topple over, i doubt we'll need any sort of support bracket between structures. Due to the size of the trailer, the current pit's are roughly 5' 10" tall (a foot and a half shorter at the minimum) and 20" wide, making it even more unlikely to topple than the roll out carts Furthermore, as long as we are smart about our organization, and put heavier bozes towards the bottom, we'll be completely fine without any sort of support beam

AniAravind, 2023-10-19:
https://cad.onshape.com/documents/3403834b07e4f3fc1af36460/w/34ad64f2a03a9e5897971e8c/e/86fa7650993cb6c5ccb1b3c1

My pit design is also made of the 80/20 1.5inx1.5in metal bars. I have labeled what each section is intended to store in the part studios of each shelf unit. I haven't decided whether to use L-brackets or shelves yet, but once that is decided I will add them in. It can store up to 16 large milwaukee boxes (I can add room for more if needed). There are shelves for power tool storage during comp, as well as drawers (must be built and added) for smaller hand tools. There are two work stations (one for electrical, one for programming) and battery storage, of course. I have also added an extra area that can be used for more storage or as a business/PR display area, and a place to store/ give away team merch+pins. The walls could either be slid into the slots in the 80/20 bars or bolted on if we require more room. The two 4.5 feet shelves that make up each side will fold and latch together during transit.

FBurkelwsd, 2023-10-26:
https://docs.google.com/document/d/1HXles1faesJuUec7-edAP0L1oUNJfDbu/edit?usp=drive_web&ouid=113250307686855979825&rtpof=true

SlavaS18, 2023-10-26:
Pit things link:
https://docs.google.com/document/d/18gDqT9lVgaOXMjqefLpzM_VAF9WaRwzx/edit?usp=sharing&ouid=103864491398126461990&rtpof=true&sd=true

FBurkelwsd, 2023-10-26:
Ben and Ted
Pit is not complete.

Need to continue collaborating, and work with ted to get final design approved

ArvickC, 2023-11-29:
On track to be reviewed by Ted on 11/29. construction on track to be started dec 6th (if approved), as in the PM preseason plan

ArvickC, 2024-02-05:
Will have an answer from @TJScher about workbenches 2/7