packofparts/frc2024 #118: Refine design based of prototype

ArvickC, 2023-09-25:


FBurkelwsd, 2023-10-11:
parts to incorporate. 
https://www.mcmaster.com/47065T356 hinges for folding the storage together and locking in place
https://www.mcmaster.com/47065T357/
https://www.mcmaster.com/47065T345/
https://www.mcmaster.com/47065T356/ 

https://cad.onshape.com/documents/3403834b07e4f3fc1af36460/w/34ad64f2a03a9e5897971e8c/e/fbe575708f3503b621353935

FBurkelwsd, 2023-12-08:
In the top left corner of the picture where the two longest modules intersect. Does the slight overlap impact technicians ability to pull materials out properly? Unable to see from the top Down view.

SanchitArora44, 2023-12-08:
Nope. the amount of overlap is exactly. 1.5", so the overlap only covers the width of the 80-20 beam over there and in no way impacts the technician's ability to take out or put in things.

TJScher, 2023-12-09:
I am happy the board has decided to stick with the L-shape design as was discussed at the beginning of this process. That is what we agreed to, and I approved in principle in the discussion on the Nov 28th with Maahimodi. Although the CAD for the entire pit was not complete the basic concept for Carts A and B were complete enough to start building them so they would be available for this season and not interfere with build. It was agreed that carts A and B would be the same length and width. On Dec 2 We started verifying the CAD to develop a cut list. We found some issues with the CAD and revised the cut list and started cutting parts. On Dec 5 when we started I was told that some of the cuts where wrong and they were recut (they weren't wrong) and the two carts were no longer the same length. The above diagram is not a compelling argument support changing length and is not what and is agreed to. If we continue down this path of change we will waste more time and material. I see us proceeding one of two ways; Maahi and I continue per our agreement, No cutting material or design changes unless approve by us, or we stop the project and wait until the entire pit is designed and approved. I expect that doing later will require us to use last years pit this year. I will plan to proceed with option one.

Maahimodi, 2023-12-11:
To start off, I thank all the mentors such as Ted and Ben that are involved in this process. I acknowledge the effort and time our mentors have taken to make sure that this project goes smoothly. I know that just like the pm board and the other members of this team, the mentors want the best for this team. 

The plan that had been approved by Ted (and Ben) was that the pit structure would consist of two shelves (A & B) of equal sizes and two Husky work benches as written in the “Finalized Pit Design” under the Board Drive. 
However, after talking to Ted about the pits yesterday, it appeared that the mentors still seemed unsure of the Husky work benches, and instead wanted to build two entire workbenches by the 80.20 design. While this may have been a communication gap, it was specifically listed that we wanted Husky Workbenches on the Finalized Pit Design Document; a document that was posted not only on slack, but on google drive as well. We also talked about the Husky workbenches being directly implanted onto the 80.20 design for 2024-2025, making them a **permanent solution**. 

As for 2023-2024, they would be the ideal standalones as they are _significantly cheaper_; a random linear bearing off the 80.20 catalog is 99 dollars just for one, and we would need two per drawer, and 9 drawers per workbench to be comparable to the husky workbench. Furthermore, building two workbenches from scratch would take significantly longer as we would have to CAD the design, order more parts, cut more 80/20, and assemble everything. As we can see, building cart A is taking much longer than expected because we need to be meticulous of every process (and for good reason of not wasting material) and measuring, recutting, and remeasuring. That being said, building the workbenches from scratch would waste valuable time that we can utilize elsewhere. While we know that the 80.20 metal extrusions are of industrial quality, Husky workbenches are also known to be of professional quality as well, with over hundreds of positive reviews. Not only does it come fully assembled (apart from casters) but has an integrated power strip that can easily power the tools we need for the robot, along with the robot batteries. Just like our red toolchest at the shop, this husky workbench can last for years without wear. While a workbench made from scratch out of 80.20 is not a bad idea, a Husky workbench would not only be more time-efficient, but less expensive, something that is worth considering, especially after the removal of 15k from our budget.

 To note, the main preface of this pit structure is to provide convenience to the technicians, programmers, and the rest of the pit crew. Our previous orientation resulted in losing 32% of total Milwaukee storage. With this new orientation, while it may seem like we’re changing up the entire design, **we are just moving the position of the workbenches and shelves, removing only 2.5 inches of length from cart B, which in the end, is just empty space**. This is preferable to buying a smaller 42-inch Husky workbench, which only has a width of 18.1 inches, which the electrical bag can barely sit on the top of. 

While all of us want the 80.20 pits to be done as fast as possible, it’s important to remember that we also want our pits to be as optimized as they can be, an idea that the PM board and the mentors both share. We believe that this design achieves that goal. The changes have been updated into CAD, along with clear descriptions of what and why changes were made, and are ready to be approved. We’re unsure of why this process would be delayed into 2024. 

As a core value of our team is being **student-driven**, we are going to continue to push for our design, but are willing to refine based on mentors’ comments. We hope the mentors feel the same and understand our concerns and willingness to work together as a team to resolve this issue.

FBurkelwsd, 2023-12-12:
"To note, the main preface of this pit structure is to provide convenience to the technicians, programmers, and the rest of the pit crew. Our previous orientation resulted in losing 32% of total Milwaukee storage. With this new orientation, while it may seem like we’re changing up the entire design, we are just moving the position of the workbenches and shelves, removing only 2.5 inches of length from cart B, which in the end, is just empty space. This is preferable to buying a smaller 42-inch Husky workbench, which only has a width of 18.1 inches, which the electrical bag can barely sit on the top of." images are definitlely needed for this. 

"As for 2023-2024, they would be the ideal standalones as they are significantly cheaper; a random linear bearing off the 80.20 catalog is 99 dollars just for one, and we would need two per drawer, and 9 drawers per workbench to be comparable to the husky workbench. Furthermore, building two workbenches from scratch would take significantly longer as we would have to CAD the design, order more parts, cut more 80/20, and assemble everything. As we can see, building cart A is taking much longer than expected because we need to be meticulous of every process (and for good reason of not wasting material) and measuring, recutting, and remeasuring. That being said, building the workbenches from scratch would waste valuable time that we can utilize elsewhere. While we know that the 80.20 metal extrusions are of industrial quality, Husky workbenches are also known to be of professional quality as well, with over hundreds of positive reviews. Not only does it come fully assembled (apart from casters) but has an integrated power strip that can easily power the tools we need for the robot, along with the robot batteries. Just like our red toolchest at the shop, this husky workbench can last for years without wear. While a workbench made from scratch out of 80.20 is not a bad idea, a Husky workbench would not only be more time-efficient, but less expensive, something that is worth considering, especially after the removal of 15k from our budget." 8020 is just one distributor, has anyone done research on others? 

"As a core value of our team is being student-driven, we are going to continue to push for our design, but are willing to refine based on mentors’ comments. We hope the mentors feel the same and understand our concerns and willingness to work together as a team to resolve this issue." As well as gracious professionalism. Imagine you have the best source in the world telling you an answer and then doing the opposite thing. This is what is happening when students ignore/ dont justify, and dont cite contradictory evidence to claims from mentors. Especially on non-shiny new piece of technology solutions. We expect data, images, and throurough research, the answer should not be made up in 10 seconds.

FBurkelwsd, 2023-12-21:
Looking into casters. Wondering how many locking swivel casters we need to have.
We could save money by not having all locking swivel casters. Here is my beginning research. We have to import the cad for alternatives for the 8020 cadd.
https://www.mcmaster.com/products/casters/creston-casters-with-rubber-wheels/ these are 60 instead of the 
https://8020.net/2338.html is the wheel we are looking to beat. 
https://shop.servicecaster.com/Heavy-Duty-Rigid-Swivel-Casters-s/1827.htm havent had time to finalize this one.
https://shop.servicecaster.com/solid-polyurethane-wheel-casters-s/2183.htm this one is half the price

ArvickC, 2023-12-28:
Cart A's 'frame' has been fully assembled and tested to fit (new) Milwaukee packouts and batteries. and parts for Cart B have been cut. Currently working on assembling the battery shelves and wood panels for Cart A. Also working with Ted to pick casters for the carts.

ArvickC, 2024-01-15:
Cart B's construction began, 'frame' and wood top are complete.

closed 2024-02-05 by ArvickC