packofparts/frc2024 #120: Design Driver/Operator Station

ArvickC, 2023-09-25:


SanchitArora44, 2023-09-25:
Sanchit's Version:
[image]
* 38" long driver station
* handles
* 3/4" thick wooden top, 1/8" allum bottom and sides
* plenty of storage area for wires and stuff
* Indented plate for secure laptop mounting
* spots for two 3D printed Xbox controller holders
* spots for 16 buttons for operator use
* Driver joysticks, if used, can be placed on the side, off the driver station.
* Cheap, easy and quick to mfg design

FBurkelwsd, 2023-10-11:
this seems like driver station should be in a different project. not under pit design. unless we want to have a place to attach this to inbetween matches

SanchitArora44, 2023-10-11:
I agree. I think the plan was to be able to store the driver station on the robot cart as many other team's do.

ArvickC, 2024-02-10:
Another design:
https://cad.onshape.com/documents/e6e526814530b6287f493f92/w/28f3ebb8da4ff0aefc9356fc