packofparts/frc2024 #121: Assemble Driver/Operator Station

ArvickC, 2023-09-25:


FBurkelwsd, 2023-10-11:
This seems to be for another project

FBurkelwsd, 2024-02-15:
How can we ensure that the ful robot wheels can be tested, while robot is sitting ontop of cart? IS this cart compatible with multiple dimensions of drive base? Are rivets and gussets the most secure way? can we go with materials we have in surplus? U channel? and other aluminum behind carts a,b,c,d.  Can we do a simulation on this to make sure we have, enough support on the handle. How can we ingrain 1294 and Pack of Parts into this cart?  Where will hands go when pushing? Where will bumpers be stored? Can we have hanging areas for tools and batteries, What is the base layer built out of, or if it can have rails for items to not slide off? Can we ensure how flexible this is based on new robot designs.  https://cad.onshape.com/documents/e6e526814530b6287f493f92/w/28f3ebb8da4ff0aefc9356fc/e/dc9647427215095ebbe336d8