packofparts/frc2024 #147: Extra swerve base design review

FBurkelwsd, 2023-10-09:
https://cad.onshape.com/documents/99c478c0c1ae8377ece0661d/w/af5196b6d6cf8b036a330bce/e/4db2bd760e06d62ea4c839c4

Take a look at the following swerve base, analyze fasteners, dimensions and spacing, material choice, and manufacturing limitations. 

Comment in the cad file using the comment button on the top left of the screen

SanchitArora44, 2023-10-09:
It tells me that i don't have permission to view this document. You might need to change the perms

adshah2, 2023-10-11:
Looks pretty good rn 👍 

Couple things though. Make sure to find a good way to mount the pdp. Right now its mounted at a slant which I think you did to prevent wire tension at certain points. Though you have to be careful to mount it as at this angle, movements will almost definitely make you lose power to robot. For PDP, its always a good practice to mount with bolts and nuts (not tape or anything like it). One idea is to create a mount where you have battery go above the PDH, like a second level. Also, the way it is mounted now, all your wires that are routed to sparks will just be hanging lose in the air.

Would recommend adding where you will mount circuit breaker and radio. If you plan on using this robot for vision testing. You will also need a network switch (as there arent enough ethernet ports on rio) which are mounted on din rails.

Since you are using MK2 Swerve, you will have absolute encoders coming out of the modules. These can get messy, would recommend figuring out where you want to route them through (and if you have to cad anything for it) as you do not have a lot of space to work with so with the added elements and all the wires, you'll be surprised how quick you run out of space for even good wire organization. 

Lemme know if you have other questions.

closed 2024-01-04 by ArvickC