packofparts/frc2024 #173: Spark Mounts

adshah2, 2023-11-28:


adshah2, 2023-11-28:
Currently have V1s of multiple mounts including on electrical board and on NEOs. @s-bitshaki is working on designing more mounts and using different tools and software to create higher quality spark mounts.

s-bitshaki, 2023-11-28:
V1 turned out great. Small adjustments needed to be made in order to secure a better fit. This successful first iteration shows great potential for applying generative design to our robot and its subsystems.

s-bitshaki, 2023-11-29:
Just put the "spark MAX mount - velcro strap" through topology optimization and currently printing prototype for testing.

s-bitshaki, 2023-11-29:
V2 of the Neo Motor to Spark Mount made with slots for zip ties, looks promising. Will also try a proto with extended legs that wrap around the neo for a better fit.

s-bitshaki, 2023-11-30:
Remade the SMAX mount to a slimmer version with the Velcro insert under the body of the spark taking less real estate. 

I then sent the mount through topography optimization reducing the simulated mass by 39.18% which is a big improvement.

s-bitshaki, 2023-11-30:
Neo-Spark Mount

I had to shorten the base between the arcs that connects directly to the Neo, the arcs were blocking the wires on the bottom. I shortened the base by 12mm and then regenerated the part through Gen Design. Many new iterations were generated and I am currently sorting through them. Once I find one I am satisfied with I will need to do the following:

1. Convert to Mesh 
2. Smooth out the mesh body and mitigate any imperfections
3. Convert to Solid Body (prismatic, NOT faceted)
4. Fix the sharp edges that were rounded due to the smoothing tool
5. Add the zip tie holes on the arc

We will be leaving the arc lengths the same as the side profile of the spark, no need for extended arcs. Waste of material and increasing overhang angle will negatively affect the print quality as well as add supports (waste of material)

s-bitshaki, 2023-11-30:
Design of the shortened Neo-Spark Mount finished and is ready for testing.

closed 2024-03-05 by adshah2