packofparts/frc2024 #177: Flash Radios

adshah2, 2023-11-29:
We need to flash all of our radios to the new 2024 software to prepare for our season.

KatsuKorr, 2023-11-30:
**CURRENT FLASHING RADIO SETUP INSTRUCTIONS:**

Materials: Ethernet cable, Ethernet to USB adapter, Radio, 12 VOLT  powersupply, unrestricted/personal laptop
			
1. Disable every network adapter except the ethernet adapter.											
2. Download JAVA runtime enviroment onto your laptop.											
3. Open onshape, open electrical files, open radio programming, open and download the FRC_Radio_Configuration file.					
4. Connect the ethernet USB adapter to your laptop.												
5. Connect the ethernet cable to the USB and the left most ethernet port on your radio (if it doesn't work use the right most port).				
6. Plug in your 12 volt power supply into a general power outlet.											
7. Connect the  power cable into your radio and the other end into your 12 volt power supply.								
8. Inside the program on the FRC file follow the instructions on the bottom left labeled "To program your wireless bridge".					
9. After completion press the main button on the program labeled "Load Firmware".									
10. This will lead you to a small list of instructions in a window, follow them accordingly.								
11. Unplug the power cable for its radio port, wait an estimated 6 seconds and plug it back into its port.							
12. Wait until a window pops up on your desktop saying "Firmware Succesfully Flashed."								
																
**WHAT WE NEED TO MAKE THE SETUP EASIER:**				
				
A designated small desktop that’s unrestricted, runs windows and has a ethernet port. This desktop should have the FRC configuration file, JAVA Runtime  pre downloaded.

TJScher, 2023-12-09:
KatsuKorr Great job documenting the process! A couple of things to add/change. First; After we downloaded the file in step 3 we checked it against the on the FRC website.  They were the same so we continued. Had they not been the same we would have downloaded the new one from the FRC website and replaced the old one in Onshape. Second: We should Load the Firmware before we program it. The flashloader may erase what we just programed. (move step 8 to the last step)

adshah2, 2023-12-13:
Still need to flash to 2024 software when released

adshah2, 2024-01-17:
All radios are flashed and ready for 2024. 
Current info is in [this spreadsheet](https://docs.google.com/spreadsheets/d/1gVPGF3MSFQXRlvdx36s0ngakO3IUibNFH0uU5BXu7Xk/edit?ouid=116590989727397826051&usp=sheets_home&ths=true)

closed 2024-01-17 by adshah2