packofparts/frc2024 #180: Battery case

adshah2, 2023-11-30:
Make a new iteration of battery case

s-bitshaki, 2023-11-30:
We will reiterate the battery case for a slimmer more functional version with the following features:

1. Plug with be snug against the body for a clip in attachment similar to putting a Milwaukee battery into its charger.
2. Analog gauge attached to the top of the case for ease of viewing voltage/battery health (if legal)
3. Nubs on the side so the batt can clip in firmly when attached (horizontally)
4. Folding handle to easily carry the battery. Folds to the side when done to save space. 
5. Sleeve on bottom of battery to make the case assembly flush 

The whole project will be optimized using Gen Design and Topography Optimization.

vkirub, 2023-12-06:
2/ Why Analog Guage as opposed to either a digital one, or something that allows for remote monitoring?

I assume this is for when the battery is not being actively used correct?

closed 2024-06-13 by adshah2