packofparts/frc2024 #184: Safeway grant analyze

FBurkelwsd, 2023-12-11:


FBurkelwsd, 2023-12-11:
[safeway grant first driaft from 22-23.pdf](https://github.com/FRC-1294/frc2024/files/13638880/safeway.grant.first.driaft.from.22-23.pdf)

We should be able to edit this or add to this grant or even submit a brand new one

jjalla, 2023-12-11:
Current Grant is for Districts. Do we want to have this grant exclusively for STEM Kits or just the team in general?

FBurkelwsd, 2023-12-13:
main part is to see the questions so we can draft our own answers.

FBurkelwsd, 2024-01-13:
How should we update this? what should we write a grant for?

FBurkelwsd, 2024-01-13:
[image]

this was submitted in 2022 and we havent heard back

FBurkelwsd, 2024-01-13:
Do we go with stem kits or an upgraded drills/tools/ materials to upgrade cnc/3d printers  housing? How about 8020 and t slot to replace our 2x4 structure in D125?