packofparts/frc2024 #204: Begin prototype CAD

ArvickC, 2024-01-08:


SanchitArora44, 2024-01-18:
Same mechanism as launcher.

SanchitArora44, 2024-01-18:
Elevator CAD V1 is finished and here's a list of Todo's based on feedback/design reviews.

Todo:
* Improve L-gussets for base and first stage (ttb ones have been known to break
* Make gusset to merge top sprocket and tube (8177 style)
* Switch to simplified MP
* Add CF spring to top of first stage to help counteract gravity
* Incorporate wire/1x1 for support+ reaction bar
* Derive drivebase tubes into elevator doc
* Have channels for rope in shooter pivot
* Reference 8177 2023 for wrist design if needed
* Add a pc plate over inverted motor area for elec/sponsors/note deflection
* Throuhbore encoder on elevator (check with programming if necessary)
* CAD in shaft collars for sprockets
* CAD in hardware like bolts and rivets
* check if falcons clips dt
*  Check why are pulley bolts on carriage so long and if they need to be that long
* Tube plug for base stage to help with sturdiness/overbuild it

SanchitArora44, 2024-01-19:
All previously listed items other than a few have been completed. The remaining items are simply design requirements for other systems/will be added later based on rest of the CAD.

Todo:
* Incorporate wire/1x1 for support+ reaction bar
* Have channels for rope in shooter pivot
* Reference 8177 2023 for wrist design if needed

For all intents and purposes (unless someone identifies an issue), the elevator CAD is finished and assembled, so please give it a final review and alert me of any issues

SanchitArora44, 2024-01-21:
After merging the elevator sub-assembly into the master-assembly i noticed a few interference issues. Screenshots of said merging issues are posted below. They're fairly easy fixes and will be addressed by the end of today

[image]

[image]

closed 2024-01-27 by ArvickC