packofparts/frc2024 #235: Construct Intake

ArvickC, 2024-01-22:


ArvickC, 2024-01-29:
lead times for intake plates are a couple days

esjyoon, 2024-02-02:
All polycarb tubing was cut for the first iteration of the intake. Hex shafts are nearly finished but need to be tapped w/ the 1/4 20 tapping bit. Custom parts that needed to be printed should be ready by 2/2.

ArvickC, 2024-02-05:
Hiccups with the printed parts, re-printed and continued progress 2/3

ArvickC, 2024-02-17:
Upcoming goals
- Mount the mdf intake with the wedge on the drivebase and test it using PWM
- Along with the pivot motor mount this should be the highest priority so we can hopefully mfg it tomorrow. Already have mounting plates cut out for intake just need go ahead on main plates
- Take a look at the current tubes and decide what needs to be recut and what is fine as is
  - if we get go-ahead on the geo re-cut and re-assemble what is necessary
  - re-wrap all the tubes (not just new ones) with tape so that there are no gaps on the surface of the tube (i don't want to be able to see any polycarb)
- Cut out and start making the 26" long polycarb roller. we'll laser cut out a testing rig for this so we can see which compression is the fastest to feed in (we want to test out 0.2", 0.15", 0.1", and 0.05"). Testing rig will also allow us to confirm the c-c for the belt

closed 2024-02-19 by ArvickC