packofparts/frc2024 #242: Cart A

FBurkelwsd, 2024-01-27:
Put L-brackets in both carts
Make holes for battery charger for MDF
Establish electrical connection (invite electrical)
Get MDF for sides and back
(Paint cart A/B with POP logo)
Place hinges on carts
Place top piece butcher block on cart B and top piece MDF on cart A
Decide which Milwaukee cases need to taken for comp
Number Milwaukee cases

FBurkelwsd, 2024-01-27:
https://cad.onshape.com/documents/7ba2ae971eba73a164f42457/w/64582b7b09eba05c730a35f0/e/2906e681d4942c5dca866066

ArvickC, 2024-02-05:
L-channels are all attached. Battery shelves are all cut out. The next steps are to bolt battery shelves (with L-channels) and the battery charger.

TJScher, 2024-02-10:
Before bolting on battery shelves, we need to add the parts to hold the batteries on to the shelves.
I suggest we assemble one prototype including the connector interface and fuse bracket before cutting and drilling the parts for the rest of the shelves. Parts required for the prototype are follows.
4 L brackets (The ones that were cut and used to define mounting holes in shelves)
1 3/4" thick wood bracket (Cut to fit using scrap, once assembled measure final size, will cut them on table saw)
1 connector bracket (Needs to be designed)
1 fuse holder bracket (Needs to be designed, needs to accommodate two Eaton BK/HTB-26I-R, see attached spec sheet.)
[eaton-htb-panel-mount-fuse-holders-data-sheet.pdf](https://github.com/FRC-1294/frc2024/files/14228406/eaton-htb-panel-mount-fuse-holders-data-sheet.pdf)
1 Anderson Power Pole connector with wires to interface to fuse holder. (wire size to be determined by max charge current, length by fuse and connector location)

ArvickC, 2024-02-17:
Split into #273 #274 #276 and #277