packofparts/frc2024 #255: Emergency Contact integration

FBurkelwsd, 2024-02-01:
Extracting information from competition forms for March and April.  and adding to our mentor emergency contact number list

vkirub, 2024-02-18:
This has been completed. @FBurkelwsd - marking this as resolved. Please file a new task if we should explore docusign or something for next year signups.

FBurkelwsd, 2024-02-18:
[image]

FBurkelwsd, 2024-02-18:
[image]

FBurkelwsd, 2024-02-18:
Anyone else see something else. Not able to access all the embedded co tact's through voice

FBurkelwsd, 2024-02-18:
Though when I go into search members pop up

FBurkelwsd, 2024-02-18:
Nicely done

closed 2024-02-18 by FBurkelwsd