packofparts/frc2024 #256: Construct bumpers for DB

ArvickC, 2024-02-05:


esjyoon, 2024-02-06:
Cut out most of the wood for bumpers 2/3

ArvickC, 2024-02-10:
Started assembly of bumpers 2/9

[image]

ArvickC, 2024-03-05:
Still need to attach cloth/fabric and team numbers. Final CAD can be found [here](https://cad.onshape.com/documents/4806c351f448d3244610d1ae/w/675b82705539b0d7ce7b0350/e/b3175b3ef1f9b7f136f14b15)

ArvickC, 2024-03-06:
Front bumpers are slightly high, and need to be lowered so the top is 7.5" off the ground to follow the guidelines. The second set of bumpers need to be constructed as well.

ArvickC, 2024-03-08:
All bumpers are done - red ones need to get wrapped

ArvickC, 2024-03-12:
Wrapped and painted by @Maahimodi

closed 2024-03-12 by ArvickC