packofparts/frc2024 #259: Second Iteration Launcher

ArvickC, 2024-02-05:


ArvickC, 2024-02-17:
- Start assembling final launcher
  - Wrap the new polycarb rollers with tape (please wrap them nicely. don't want to see any gaps)
  - finish process the plates cut last meeting
  - Take launcher (only launcher not wrist) parts from the mdf version and begin assembling final version with those
  - get stuff needed for flywheels cut and begin that assembly
- Finish testing the c-c's for the motor mounts with the abs encoder so CAD can be adjusted if needed or we can manufacture the platesometry finalized in the CAD.

esjyoon, 2024-02-21:
Polycarb roller parts will be replaced with their nylon counterparts Wed 2/21. Rollers will also need to be grip-taped. Launcher is waiting on 40A 3" compliant flywheels.

ArvickC, 2024-02-24:
Mechanically complete. @FRC-1294/electronics will continue wiring

ArvickC, 2024-02-29:
Problems with the belt skipping on the wrist. Ordered custom gears to fix belt skippage and they are arriving 3/1.

ArvickC, 2024-03-05:
Custom gears ordered and attached

closed 2024-03-05 by ArvickC