packofparts/frc2024 #260: Second Iteration Elevator

ArvickC, 2024-02-05:


esjyoon, 2024-02-06:
New tubing was cut out and made square 2/5, will need to be run through the CNC

esjyoon, 2024-02-14:
Second iteration elevator finished and mounted as of 2/13, still need to switch out for hooks and final wrist/launcher system (ideally by 2/14)

ArvickC, 2024-02-17:
- add and wire cameras
- finish final wiring in general (cams, motors, energy chain, etc)
- If the CNC is free i'll cut out the allum sprocket/tube mounts for the top and the pocketed elevator mounts for the backup elevator tmrw and then mech people can post process and mount it if we have enough people.
- Cut the hex shafts on the elevator motor down to be flush with the gears (we don't have the clearance to allow for longer than necessary shafts)

ArvickC, 2024-02-29:
Mounting hooks onto elevator 2/29
Mounting new plates on elevator 2/29

closed 2024-03-05 by ArvickC