packofparts/frc2024 #268: Auburn

FBurkelwsd, 2024-02-10:
Have requested quotes for 3 passenger vans and a bus 
Who is going?
How will they get there?
Food?

wygrangea, 2024-02-10:
We have multiple drivers already for Auburn, Ben and I do not plan to drive for the event as we are working it and staying down there with family.
I assume everyone able to go is going, and @jjalla has a spreadsheet, maybe it could be linked here? I can arrange food delivery if you are spending money on it or there are concessions. Maybe Wednesday night we can discuss what food you'd want.

FBurkelwsd, 2024-02-13:
Can we update the drivers on https://docs.google.com/spreadsheets/d/1FrxNh6sAQfZVmEdy6gyxVXYIZcIUpU_s/edit#gid=881354543 out master form, go to the right for bus seats and carpools

closed 2024-04-17 by ArvickC