packofparts/frc2024 #269: Sammamish hs

FBurkelwsd, 2024-02-10:
Have requested quotes for 3 passenger vans and a bus 

Who is going?
How will they get there?
Food?

wygrangea, 2024-02-10:
I thought we weren't bussing here - @jjalla  has a spreadsheet of who needs a ride. I will not be helping with transport. For food we are hoping to have food trucks, or can order in food for the team or have people bring lunch. I can share what I do for 2412 Wednesday.

FBurkelwsd, 2024-02-13:
https://docs.google.com/spreadsheets/d/1FrxNh6sAQfZVmEdy6gyxVXYIZcIUpU_s/edit#gid=881354543 is the spread sheet we need to update drivers and confirm with all that they have a way to get to comp

jjalla, 2024-02-14:
These are the students that need a carpool:
Aaditya Agnihotri
Aaryan Sood
Aditya Gajula
Aditya Shah
Alex  rojas
Alex Taillefer
Angus tian
Ankit Malhotra
Arisha Gupta
Ben uttecht
Bogdan mischie
Camila Figueira
Cooper chi
Elliott Webb
Ethan Yoon
Gayathri Arvindh
Kritish Alli
Maahi Modi
Sarathkiran Saravanan
Shivani Vivekanand
Siddhant Jena
Sushanth Reddy Malireddy
Teja koripella
Tiana Chi
Trinabh Gupta
Vaibhav Satish
Vineeth Rajesh
Winston Luo

jjalla, 2024-02-14:
Drivers Saturday Morning:
Vivek(6)
Cathy Godfrey(5)
Jitin Sood(5)
Frederick Burke(4)
Jennifer Rojas(4)
Charu Arora(4)
Carlos Figueira(4)
chiahsin chang(5)
Total Capacity: 37 students

Drivers Saturday Afternoon:
Vivek(6)
Cathy Godfrey(5)
Jitin Sood(5)
Frederick Burke(4)
Jennifer Rojas(4)
Carlos Figueira(4)
chiahsin chang(5)
Total Capacity: 33 students

Drivers Sunday Morning:
Vivek(6)
Cathy Godfrey(5)
Jitin Sood(5)
Frederick Burke(4)
Jennifer Rojas(4)
Carlos Figueira(4)
chiahsin chang(5)
Total Capacity: 33 students

Drivers Sunday Afternoon:
Vivek(6)
Cathy Godfrey(5)
Frederick Burke(4)
Jennifer Rojas(4)
Charu Arora(4)
Carlos Figueira(4)
chiahsin chang(5)
Total Capacity: 32 students

jjalla, 2024-02-14:
In my opinion we have plenty of drivers and do not need rental vans or a bus.

wygrangea, 2024-02-14:
That looks good, let's confirm in the spreadsheet tomorrow.

closed 2024-04-17 by ArvickC