packofparts/frc2024 #271: Houston

FBurkelwsd, 2024-02-10:
Who is going?
How will they get there?
Food?

wygrangea, 2024-02-10:
For clarification:
1. When do you want this info?
2. Is it for admins or just so we have direction?
3. who goes should be decided by team metrics, how many students do you want to take if we qualify? The smaller the group, the better. 
4. We will book airfare  (likely on United), hotel will be booked by Cvent (I actually have 2 hotel requests in, so we can see if you need one)
5. I can arrange daily lunch delivery or students can be on their own. I recommend deciding on dinner in chaperone groups or having ezcater.com deliver to the hotel. I have a whole setup we can go through if we qualify for DCmps.

FBurkelwsd, 2024-02-13:
https://docs.google.com/spreadsheets/d/1FrxNh6sAQfZVmEdy6gyxVXYIZcIUpU_s/edit#gid=881354543

All of this is for being able to make a reservation and making sure we dont get double charged. admin wanted this months ago, so we were right submitting a rough draft that we are now confirming

We will want to have a working number definitely before dcmp, we will have a feel how realistic this is by then.

closed 2024-04-17 by ArvickC